tusen-ai/SST #132: tranning problem

20210726, 2023-08-15:
**at Step 5: Begin training,here is error:**
2023-08-15 19:05:25,999 - mmdet - INFO - workflow: [('train', 1)], max: 24 epochs
INFO:mmdet:workflow: [('train', 1)], max: 24 epochs
2023-08-15 19:05:26.085644: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcudart.so.11.0
Traceback (most recent call last):
  File "tools/train.py", line 230, in <module>
    main()
  File "tools/train.py", line 220, in main
    train_model(
  File "/waymo/SST/mmdet3d/apis/train.py", line 41, in train_model
    train_detector(
  File "/root/anaconda3/envs/ctrl/lib/python3.8/site-packages/mmdet/apis/train.py", line 170, in train_detector
    runner.run(data_loaders, cfg.workflow)
  File "/root/ctrl/mmcv/mmcv/runner/epoch_based_runner.py", line 127, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/root/ctrl/mmcv/mmcv/runner/epoch_based_runner.py", line 47, in train
    for i, data_batch in enumerate(self.data_loader):
  File "/root/anaconda3/envs/ctrl/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 517, in __next__
    data = self._next_data()
  File "/root/anaconda3/envs/ctrl/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1199, in _next_data
    return self._process_data(data)
  File "/root/anaconda3/envs/ctrl/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1225, in _process_data
    data.reraise()
  File "/root/anaconda3/envs/ctrl/lib/python3.8/site-packages/torch/_utils.py", line 429, in reraise
    raise self.exc_type(msg)
KeyError: Caught KeyError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/root/anaconda3/envs/ctrl/lib/python3.8/site-packages/torch/utils/data/_utils/worker.py", line 202, in _worker_loop
    data = fetcher.fetch(index)
  File "/root/anaconda3/envs/ctrl/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/root/anaconda3/envs/ctrl/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/root/anaconda3/envs/ctrl/lib/python3.8/site-packages/mmdet/datasets/dataset_wrappers.py", line 151, in __getitem__
    return self.dataset[idx % self._ori_len]
  File "/waymo/SST/mmdet3d/datasets/waymo_tracklet_dataset.py", line 284, in __getitem__
    data = self.prepare_train_data(idx)
  File "/waymo/SST/mmdet3d/datasets/waymo_tracklet_dataset.py", line 209, in prepare_train_data
    input_dict = self.get_data_info(index)
  File "/waymo/SST/mmdet3d/datasets/waymo_tracklet_dataset.py", line 139, in get_data_info
    trk.set_type(self.cat2id[trk.type_name], 'mmdet3d')
KeyError: 'Pedestrian'

Killing subprocess 2618
Traceback (most recent call last):
  File "/root/anaconda3/envs/ctrl/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/root/anaconda3/envs/ctrl/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/root/anaconda3/envs/ctrl/lib/python3.8/site-packages/torch/distributed/launch.py", line 340, in <module>
    main()
  File "/root/anaconda3/envs/ctrl/lib/python3.8/site-packages/torch/distributed/launch.py", line 326, in main
    sigkill_handler(signal.SIGTERM, None)  # not coming back
  File "/root/anaconda3/envs/ctrl/lib/python3.8/site-packages/torch/distributed/launch.py", line 301, in sigkill_handler
    raise subprocess.CalledProcessError(returncode=last_return_code, cmd=cmd)
subprocess.CalledProcessError: Command '['/root/anaconda3/envs/ctrl/bin/python3', '-u', 'tools/train.py', '--local_rank=0', 'configs/ctrl/ctrl_veh_24e.py', '--launcher', 'pytorch', '--no-validate']' returned non-zero exit status 1.

**I followed the CTRL_instructions.md, use part of waymo-dataset, and ignore step2。here should be a configuration file to fix this problem, but I couldn’t find it.**

Abyssaledge, 2023-08-16:
The key should not be 'Pedestrian' since you use the vehicle config. I need the command and adopted config for checking.

20210726, 2023-08-16:
> The key should not be 'Pedestrian' since you use the vehicle config. I need the command and adopted config for checking.

train command is :bash tools/dist_train.sh configs/ctrl/ctrl_veh_24e.py 1 --no-validate

and fsd_base_vehicle.yaml is:
`# modified from old file "fsd_pastfuture_vehicle"

# detection result in waymo bin format
bin_path: ./data/waymo/waymo_format/train_gt.bin 
val_bin_path: ./data/ctrl_bins/validation/fsd_base_vehicle_val.bin 
data_root: ./data/waymo/tracklet_data
exist_ok: False
# change the split for your need
split: training
# split: val
# split: test
type: [1,] # 1:vehicle 2:pedestrian 4: cyclist

box:
  extra_width: 1

selection:
  mode: random
  size: 1.0

candidate:
  affinity_thresh: 0.5`

Abyssaledge, 2023-08-16:
Why do you use `train_gt.bin` in the YAML file?

20210726, 2023-08-16:
> Why do you use `train_gt.bin` in the YAML file?

i think train_gt.bin is detection result in waymo bin format。so i shoud run （Step 2: Use ImmortalTracker to generate tracking results in training split (bin file format)） first，then use bin file generated in step 2 to train model？

Abyssaledge, 2023-08-16:
No, train_gt.bin contains the ground-truth information on training set. What you need here is the proposals on training set.
So do not change `bin path` to `train_gt.bin`, only change the split to `training` and use 'fsd6f6e_vehicle_full_trainset.bin' if you want to generate training data.

20210726, 2023-08-22:
I'm still confused  at how to generate 'fsd6f6e_vehicle_full_trainset.bin'. I ran step2，and get mot result in npz format。
[image]

Abyssaledge, 2023-08-22:
Please reopen this issue if you need further discussion.

closed 2023-08-22 by Abyssaledge

20210726, 2023-08-22:
1.prepare waymo data(use part of waymo dataset)
1.1 use my python script to generate train.txt val.txt test.txt and idx2timestamp.pkl idx2contextname.pkl 
Then cp train.txt val.txt test.txt to ./data/waymo/kitti_format/ImageSets/
cp idx2timestamp.pkl idx2contextname.pkl to ./data/waymo/kitti_format/
1.2 python tools/create_data.py --dataset waymo --root-path ./data/waymo/ --out-dir ./data/waymo/ --workers 128 --extra-tag waymo
[图片]
Step 1: Generate train_gt.bin once for all. (waymo bin format).
python ./tools/ctrl/generate_train_gt_bin.py
generate file 'train_gt.bin'

[图片]
python ./tools/ctrl/extract_poses.py
Generate file context2timestamp.pkl and pose.pkl
[图片]

20210726, 2023-08-22:
@Abyssaledge

JayYangSS, 2023-10-17:
step 2, I think you shouldn't use train_gt.bin: 
```bash preparedata/waymo/waymo_convert_detection.sh ~/dataset/waymo/waymo_format/train_gt.bin CTRL_FSD_TTA```

you need use base detector to  generate prediction result