tusen-ai/SST #133: What is the order to install the environment, if want to use CTRL

atlantins, 2023-08-16:
i follow the order like this:
1.https://github.com/tusen-ai/SST/blob/main/docs/overall_instructions.md
    install pytorch mmcv mmdet mmsegmentation mmdet3d
    then pip install -e . 
2.https://github.com/tusen-ai/SST/blob/main/docs/CTRL_instructions.md#step-2-use-immortaltracker-to-generate-tracking-results-in-training-split-bin-file-format
   (2)https://github.com/Abyssaledge/ImmortalTracker-for-CTRL    run_mot.sh
i really need a specific code running instruction to follow, and need enviroment installation order

Abyssaledge, 2023-08-16:
The order you follow is right. Did any errors occur?

atlantins, 2023-08-17:
1:
can you give me a scrennshot in ./data, i am confused the structure, and https://github.com/open-mmlab/mmdetection3d/blob/1.0/docs/getting_started.md,the link you give for mm3d to getting started wasn't existed, So I don't know if the dataset I'm using in mmdet3d v1.1.0 is useful for CTRL
i was confused about should we both use kiit_format and waymo_format,or just kitti_format?

2.
$ python ./tools/ctrl/extract_poses.py
Traceback (most recent call last):
  File "./tools/ctrl/extract_poses.py", line 31, in <module>
    idx_str = info['point_cloud']['velodyne_path'].split('/')[-1].split('.')[0]
TypeError: string indices must be integers

3.
https://github.com/tusen-ai/SST/blob/main/docs/CTRL_instructions.md
if i  follow the instructions, should i  use waymo_format data?
my folder like this ./data/waymo_format
(1)bash preparedata/waymo/waymo_preparedata.sh  ./data/waymo
(2)bash preparedata/waymo/waymo_convert_detection.sh ./data/waymo_format/gt.bin cp  
or  bash preparedata/waymo/waymo_convert_detection.sh ./data/kitti_format/train_gt.bin cp  
(3) run_mot.sh

atlantins, 2023-08-17:
when i run_mot.sh
$ sh run_mot.sh
Tracking time cost: 0.22504878044128418s
evaluation/waymo/pred_bin.py:195: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  config = yaml.load(open(args.config_path, 'r'))
Converting TYPE vehicle into WAYMO Format
0it [00:00, ?it/s]
Merging different object types

Abyssaledge, 2023-08-17:
No, you cannot use the dataset from mmdet3d v1.1.10, because the definition of box is different. Please use our code to generate the data. The point cloud data can be shared between different versions of mmdet3d (with minimal modifications), but the pickle info file needs to be regenerated.

Abyssaledge, 2023-08-17:
If tracking time costs less than 1 second, there must be errors in the tracking process and the main body is skipped. And the errors would not be thrown due to the multi-process. So change it to single-process for debugging.

atlantins, 2023-08-18:
so can i use python tools/create_data.py waymo --root-path --out-dir --workers 128 --extra-tag waymo in ./SST/ to generate data,or i should use mmdet3d v0.15.0 to generate data

Abyssaledge, 2023-08-22:
Yes, you could use ‘tools/create_data’ to generate data. And our repo is exactly v0.15.0.
Please reopen this issue if you need further discussion, or I may miss your posts due to too many issues.

closed 2023-08-22 by Abyssaledge