tusen-ai/SST #135: About the dimensions of the box

hutao568, 2023-08-17:
I'm not sure whether the size dimension of the box should be l, w, h or w, l, h; when you initialize, you use x_size, y_size, z_size; but when you crop the point cloud, you use w, l , h, ry; I want to know which format should be given when the box is initialized; (here w refers to width, h refers to height)

the code for init
https://github.com/tusen-ai/SST/blob/main/mmdet3d/core/bbox/structures/base_box3d.py#L19
the code for crop point cloud
https://github.com/tusen-ai/SST/blob/main/mmdet3d/ops/roiaware_pool3d/points_in_boxes.py#L6

Abyssaledge, 2023-08-18:
Always follow this format: https://github.com/tusen-ai/SST/blob/main/mmdet3d/core/bbox/structures/lidar_box3d.py#L15C16-L15C16 when you call the box-related torch ops.

Abyssaledge, 2023-08-18:
Special reminder: the boxes in waymo_infos_xxx.pkl is not in that format, but they will be converted to the LiDARInstance3DBbox after data pipeline.

Abyssaledge, 2023-08-22:
Please reopen this issue if you need further discussion.