tusen-ai/SST #140: Question about the table5 in FSD v2

LHY-HongyangLi, 2023-08-28:
Hi @Abyssaledge, FSD v1 and FSD v2 are really nice works. But I'm little confused about the results in Table5 of the v2 version. 

The results of VoxelNext on the validation set of nuScenes are 60.5mAP | 66.6NDS, but there are 63.5mAP and 68.7NDS in the table. Are there any differences in the experiment setting compared with the official results of VoxelNext?

Abyssaledge, 2023-08-28:
For fair comparison, I report the best performance of VoxelNeXt, which is provided by the author @yukang2017 here: https://github.com/BraveGroup/FullySparseFusion/issues/1

LHY-HongyangLi, 2023-08-28:
Thank you for your gently reply. I think it could be better to illustrate that the results of VoxelNext are "VoxelNext[1]+ LargeKernel3D[2]" in the caption of Table 5 to avoid confusion.

[1] VoxelNeXt: Fully Sparse VoxelNet for 3D Object Detection and Tracking
[2] LargeKernel3D: Scaling up Kernels in 3D Sparse CNNs

Abyssaledge, 2023-08-28:
Thanks for your attention and reminding me. I will update this information in the next version after checking with the author of VoxelNeXt.

LHY-HongyangLi, 2023-08-28:
OK, thank you for your timely reply and I'll close this issue.

closed 2023-08-28 by LHY-HongyangLi

yukang2017, 2023-08-29:
Hi @LHY-HongyangLi ,

To be clear, the result I provided [here](https://github.com/BraveGroup/FullySparseFusion/issues/1#issuecomment-1521093896) is just the VoxelNeXt actually, not combined with LargeKernel3D. 

“this follows "LargeKernel3D in the Table 1" for training settings”, means that I followed the training settings, LR, epochs, weight decays... but did not use LargeKernel3D for network.

The reason why it is better than the official VoxelNeXt paper is that I disabled the gt-sampling in the last training epochs in this 63.5mAP result.

LHY-HongyangLi, 2023-08-29:
@yukang2017, 
Thank you for your clarification.