tusen-ai/SST #141: dataset convert issue and how to generate tracking bin

atlantins, 2023-08-29:
[image]
[image]

When I generate official waymoopen dataset kitti_format , I often generate a directory larger than 2.8T in training folders and raise error. This is the directory generated by my training folder, and I have deleted some images. Our algorithm requires at least which folders(Can I annotate the code that generates images or can i remove images?). I don't know if you have encountered this problem

and if i have my own waymo-open dataset （make my  own waymomini）,how to generate "The detection results should be in official WOD submission format(.bin)" , through mmdetection to inference traning scenes to generate bin?how do you get it?Can you share your running command with me,
Thanks!

atlantins, 2023-08-29:
or can i use the latest version, only regenerate the pkl file ,just like you said in here https://github.com/tusen-ai/SST/issues/133

Abyssaledge, 2023-08-29:
Yes, the point cloud data can be reused with small modifications, while the pkl file needs to be regenerated.
Note the point cloud generated by the latest version has a different channel definition (such as x, y, z, intensity ...).
If I recall correctly, the first five dimensions are (x, y, z, intensity, elongation ), which is the same as the old data.
Anyway, you should check the data channel before starting experiments.

Abyssaledge, 2023-08-29:
You can get the bin file by inference with our tracking or detection algorithms.
You could refer to [this](https://github.com/tusen-ai/SST/blob/main/mmdet3d/datasets/waymo_dataset.py#L715) to see how to write boxes into bin file.

atlantins, 2023-08-31:
> Yes, the point cloud data can be reused with small modifications, while the pkl file needs to be regenerated. Note the point cloud generated by the latest version has a different channel definition [...]

https://github.com/open-mmlab/mmdetection3d/pull/1759/files，I have found the reason. This is a very small change that has been modified in the new version of mmdetection. You can make some modifications

closed 2023-09-06 by atlantins