tusen-ai/SST #142: How to visulize our detection results?

atlantins, 2023-09-01:
I have obtained the final inference result. bin, how to visualize it？ Thanks！

atlantins, 2023-09-01:
[image]
In my opinion, I added -- show to the final test, but I encountered a problem in the quickstart step and ran it here/ Configure.sh is the directory where it should be located, and the file cannot be found here

atlantins, 2023-09-01:
And there is some questions? 
1.If i want three class ctrl results how can i get it ?  (1) Base detector should give three class results in the one forward ? (2) or i just need three different class base detetor to generate results, and immortaltracker will fuse the results in pred  (3) what about the ctrl? three class or one class?
2. the result i get low map before removing ,is that normal 
[image]

atlantins, 2023-09-01:
> And there is some questions? 1.If i want three class ctrl results how can i get it ? (1) Base detector should give three class results in the one forward ? (2) or i just need three different class [...]

and can you share the single class traning config?

Abyssaledge, 2023-09-01:
I updated the visualization scripts right now. See `./tools/vis/show_bin.py`

Abyssaledge, 2023-09-01:
> And there is some questions? 1.If i want three class ctrl results how can i get it ? (1) Base detector should give three class results in the one forward ? (2) or i just need three different class [...]

Sorry, I do not really understand these questions. Feel free to use Chinese and please be more specific.

atlantins, 2023-09-04:
好的，要是这么几个问题
（1）比如说我们得到最后的检测结果的时候，是根据三个class得到的检测结果拼接得来的么？再具体一点的话，就是我的base detetor需要在一个类别上训练还是三个类别上训练，我的ctrl detetor需要在一个类别上训练还是三个类别上（询问这个问题的原因，主要是看云盘中提供的pth都是不同类别分开训练的)
（2）云盘中提供的fsd_base_vehicle_val.bin是什么的结果，是经过ctrl后的最终结果，还是base detetor的结果，fsd6f6e_vehicle_full_train是ctrl的最终结果还是base detector的结果？(此外，能够提供base detector的配置文件和pth么，可以训练自己的fsd进行验证）
（3）不知道如何验证精度，比如说我上述的图片给的那样，运行dist_test.sh后得到的map结果一般是多少比较符合，运行remove_empty_box后得到map结果一般是多少比较合适

Abyssaledge, 2023-09-06:
(1) base detector 一起训练，ctrl分别训练，当然ctrl经过简单改动也可以多类别一起。
(2) 这俩都是base detector经过tracking的结果。
(3) instruction里面写了预期性能的。
(4) 是的。
(5) 这些都是instruction里的步骤得到的（step3 & step4）。

atlantins, 2023-09-07:
> I updated the visualization scripts right now. See `./tools/vis/show_bin.py`

不知道你们之前是否有可视化连续帧的vis tool，如论文所说，一旦检测就不会丢失，不知道如何可视化一段连续帧的检测结果，而不是只有一张某一帧结果的检测图片

closed 2023-12-22 by Abyssaledge