tusen-ai/SST #144: Directly inference with my own dataset for CTRL

Eddiesyn, 2023-09-03:
I would like to directly inference the pretrained model of CTRL on my own dataset. But i found that the model input point cloud has intensity and elongation as its additional feature input. I'm wondering 
1. is there a model version of not using elongation or intensity as input; or such ablation studies showing the impact of it on the final result;
2. is there a way of hacking a dummy elongation into the input (like always use 0, or any valid constant option, since the elongation input dimension is transformed with a tanh function). Just faking that all point cloud has same elongation.

Abyssaledge, 2023-09-04:
You can remove intensity and elongation, which leads to a minimal performance drop. However, you need to retrain the model.
We do not have off-the-shelf pretrained models ignoring intensity and elongation.

Eddiesyn, 2023-09-04:
> 

Thanks for the quick rely!

closed 2023-09-04 by Eddiesyn