tusen-ai/SST #149: How to remove the false detection?

rockywind, 2023-09-06:
Hi, 
Why the nms not remove the false detection?
[image]
This is my config.
[image]

atlantins, 2023-09-07:
How you get the visulization photo like the first one ？

eagle-chase, 2023-09-08:
The default algorithm used should be 3diou_ NMS, I think other NMS methods can be used for large targets or the IoU threshold can be lowered

Abyssaledge, 2023-09-08:
Because their IoU is less than 0.25.

rockywind, 2023-09-08:
@Abyssaledge 
I think it need a nms between multi-class.
The white is the truck and the pink is the bus.
Their IOU is very large.

closed 2023-12-22 by Abyssaledge