tusen-ai/SST #150: Intensity scaling in FSDv2

ArseniuML, 2023-09-06:
I want to test FSDv2 with downloaded Waymo weights on my point clouds.
However, my LIDAR outputs intensity as UINT8 (0-255), and in Waymo data intensity maximum in various frames can be as large as ~33000  or ~80000. If I feed my data to FSDv2 as is, it produces garbage as prediction results...
How do I have to transform intensity from my datasets?

ArseniuML, 2023-09-06:
The same question about feeding my data to FSDv2 with Argo2 weights.

Abyssaledge, 2023-09-08:
We normalize intensity here: https://github.com/tusen-ai/SST/blob/main/mmdet3d/models/detectors/single_stage_fsd.py#L287 Sorry for the hard coding.
I suggest training a model without intensity, only taking xyz as input. 
Otherwise, even if you normalize the intensity to a similar range, the intensity distribution of WOD and your data may still be different.

ArseniuML, 2024-09-04:
I don't think there's much point in applying tanh to intensity (especially non-normalized), because we end with degenerated distributions.