tusen-ai/SST #153: How to improve the robustness of fsdv2 algo

eagle-chase, 2023-09-08:
In the FSDv2 algorithm, distant targets may have very few points in the point cloud. If these points are classified as background, it can result in missed detections. How can we improve the detection accuracy of distant targets? For close-range targets, false positives are often encountered, such as mistaking objects like guardrails、trees for vehicles. Is there a way to reduce false positives and missed detections in algorithm design? Does the occurrence of these false positives and missed detections primarily depend on the accuracy of the segmentation network?

eagle-chase, 2023-09-08:
I also want to know if it is a feasible method to filter the ground truth boxes based on the number of point clouds in the ground truth boxes, such as filtering out boxes with less than 5 point clouds.

eagle-chase, 2023-09-08:
If I want to increase the depth of the model, which part of the model do you think would be best to modify?

Abyssaledge, 2023-09-08:
1. Segmentation theoretically has an influence on detection accuracy. However, it seems fine in FSDv2. If a distance object is completely classified as background, then I guess it is also difficult for other detectors to detect the object. 
2. Of course you can filter these boxes. Just calculate the number of contained points. It seems straightforward.
3. It is tricky. I think the current backbone is large enough.

closed 2023-09-12 by eagle-chase