tusen-ai/SST #156: When I follow the tutorial to process training data

atlantins, 2023-09-13:
I try to reproduce the results.When I follow the tutorial to process training data
I have trained fsdv2 as my  base detetor(configs/fsdv2/fsdv2_waymo_2x.py),but when i use immortaltracking,

python3 evaluation/waymo/pred_bin.py --name $name --det_name $det_name --config_path $config --split $split --obj_types $obj_types --no-eval

i got the following text:

$ python3 evaluation/waymo/pred_bin.py --name immortal --det_name fsdv2_reproduce --config_path configs/waymo_configs/immortal_for_ctrl_keep10.yaml --split training --obj_types vehicle --no-eval
evaluation/waymo/pred_bin.py:195: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  config = yaml.load(open(args.config_path, 'r'))
Converting TYPE vehicle into WAYMO Format
100%|████████████████████████████████████████████████████████████████████████████████████████████████| 798/798 [1:20:35<00:00,  6.06s/it]
Traceback (most recent call last):
  File "evaluation/waymo/pred_bin.py", line 208, in <module>
    main(args.name, obj_type, result_folder, args.raw_data_folder, output_folder, args.output_file_name, config)
  File "evaluation/waymo/pred_bin.py", line 113, in main
    f.write(objects.SerializeToString())
ValueError: Message waymo.open_dataset.Objects exceeds maximum protobuf size of 2GB: 3119119940

Your file size is 768M，
here is the mot_results size
/ImmortalTracker-for-CTRL/mot_results/waymo/training/immortal_fsdv2_reproduce [13:44:57] 
$ du -sh *|sort -h
8.0K    bin
7.0G    summary

i add split train in this two files 
https://github.com/Abyssaledge/ImmortalTracker-for-CTRL/blob/main/preparedata/waymo/ego_info.py and https://github.com/Abyssaledge/ImmortalTracker-for-CTRL/blob/main/preparedata/waymo/time_stamp.py   

I have some questions about it ?
can you provide the basic detetor training config with FSD? or  can i use fsdv2 as the basic detetor? 
how can we generate the base detetors bin? i changed the cfg.data.test use 'data/waymo/kitti_format/waymo_infos_train.pkl' to replace 'ann_file='data/waymo/kitti_format/waymo_infos_val.pkl' , and run dist_test.sh

test=dict(
        type='WaymoDataset',
        data_root='data/waymo/kitti_format/',
        ann_file='data/waymo/kitti_format/waymo_infos_train.pkl',
        split='training',
        pipeline=[

Abyssaledge, 2023-09-17:
For training data, there is no need to conduct forward or backward extensions. Otherwise, the bin file may reach the size limit.

atlantins, 2023-09-20:
这里我有点不太理解，你能讲讲具体是哪一步需要修改么
1.我的理解是，https://github.com/tusen-ai/SST/blob/main/docs/CTRL_instructions.md,还是得通过step2，只是使用的是configs/immortal_for_ctrl_keep10.yaml，这样就可以跳过forward或者backward对么？我就是卡在python3 evaluation/waymo/pred_bin.py 上
2.我想知道你们是如何得到一开始推理的结果的（我的理解是对traning的文件用basedetetor推理一遍对么），我得到的bin文件有1.53G（我的方式：use 'data/waymo/kitti_format/waymo_infos_train.pkl' to replace 'ann_file='data/waymo/kitti_format/waymo_infos_val.pkl' , and run dist_test.sh ），你们是用什么命令推理得到的？你们是使用SST2/configs/fsd/fsd_waymoD1_1x.py训练的基础检测头么，能提供base_detetor的训练config么

Abyssaledge, 2023-09-20:
In the tracking config, you should set `max_time_since_update` to 0 for training data generation. Otherwise, you will get an output with huge size.

atlantins, 2023-09-20:
不知道对于第二个问题，能提供官方版本的训练配置文件么，这边想自己复现整个流程来评估，感觉这一步是不是对结果影响会挺大的