tusen-ai/SST #158: “__init__() takes from 5 to 9 positional arguments but 39 were given”

rockywind, 2023-09-14:
Hi， thank you for sharing the great work!
I met the error below when I run the script ```` python ./tools/ctrl/generate_candidates.py ./tools/ctrl/data_configs/fsd_base_vehicle.yaml --process 8````
[image]

Abyssaledge, 2023-09-17:
Sorry for the late reply. Have you solved this problem? And have you made any modifications to the code?

rockywind, 2023-09-17:
Hi, I have not solve it. I  follow the guide from the [link](https://github.com/tusen-ai/SST/blob/main/docs/CTRL_instructions.md).
[image]
The data was download from the google drive you provide.
I am confuse that what's the cls function.

rockywind, 2023-09-17:
This is the fsd_base_vehicle.yaml config.
````
bin_path: /SHFP12/xiaoquan.wang/01_bev/SST/01_work_dir/fsd6f6e_vehicle_full_train.bin 
val_bin_path: /SHFP12/xiaoquan.wang/01_bev/SST/01_work_dir/fsd_base_vehicle_val.bin 
data_root: /SHFP12/xiaoquan.wang/01_bev/SST/data/public_datasets/waymo_bev_ctrl/

exist_ok: False
# change the split for your need
# split: training
split: val
# split: test
type: [1,] # 1:vehicle 2:pedestrian 4: cyclist

box:
  extra_width: 1

selection:
  mode: random
  size: 1.0

candidate:
  affinity_thresh: 0.5
````
This is my generate_candidates.py config.
````
parser = argparse.ArgumentParser()
parser.add_argument('--config', default="/SHFP12/xiaoquan.wang/01_bev/SST/tools/ctrl/data_configs/fsd_base_vehicle_my.yaml",type=str)
parser.add_argument('--gt-bin-path', type=str, default='/SHFP12/xiaoquan.wang/01_bev/SphereFormer/data/public_datasets/waymo/waymo_format/train_gt.bin')
parser.add_argument('--process', type=int, default=1)
args = parser.parse_args()
````

Abyssaledge, 2023-09-22:
The `cls` is LiDARTracklet class itself. So the input arguments should be consistent with its `__init__` func. 
Could you please show what arguments are passed into the __init__?

rockywind, 2023-09-22:
感谢您的帮助，我重新走一遍流程，现在解决了。

closed 2023-09-22 by rockywind