tusen-ai/SST #160: Speedup FSDv2 via perceprion range shrinking

ArseniuML, 2023-09-20:
FSDv2 with pre-trained Argoverse2 weights seems to get good metrics on our dataset. However, our Lidar has perception range +-100 m. Do I can speedup FSDv2 via perception range shrinking?
Would it be sufficient to set perception range like this:

```
point_cloud_range = [-102.4, -102.4, -3.2, 102.4, 102.4, 3.2]
virtual_voxel_size=(0.4, 0.4, 0.4) #(512, 512, 16)
```

Abyssaledge, 2023-09-20:
Yes, you can directly change the pc_range to this one.
However, since FSDv2 is fully sparse, if there are no points beyond 100m in your dataset, I don't think it will be faster after shrinking the range.
Anyway, you could have a try. If you have any problems after the modification, feel free to post new replies or issues.

ArseniuML, 2023-09-21:
Perception range shrinking seems not to speedup the model. Are there other settings in config I can play with in order to speedup FSDv2?

Abyssaledge, 2023-09-22:
The most straightforward way is to reduce the size of backbone, and the mixer. 
For example in this [config](https://github.com/tusen-ai/SST/blob/main/configs/argo2/argo_onestage_12e.py), we adopt a very lightweight backbone, you could try it on FSDv2.

closed 2023-12-22 by Abyssaledge