tusen-ai/SST #162: Deploy FSDv2 on TensorRT

ArseniuML, 2023-09-22:
I want to deploy FSDv2 on TensorRT, but it seems that I need an MMDeploy config to do this.

Here is an example config from MMDeploy docs:

```
backend_config = dict(
    type='tensorrt',
    common_config=dict(
        fp16_mode=False, max_workspace_size=1 << 30),
    model_inputs=[
        dict(
            input_shapes=dict(
                input=dict(
                    min_shape=[1, 3, 512, 1024],
                    opt_shape=[1, 3, 1024, 2048],
                    max_shape=[1, 3, 2048, 2048])))
    ])
```

What min_shape, opt_shape and max_shape do I have to paste into this config? It seems, that FSDv2 receives an unbounded list of points as input...

Abyssaledge, 2023-09-23:
I don't think you can directly deploy FSDv2 with TensorRT, which doesn't support Sparse Convolution ...

ArseniuML, 2023-09-25:
OK, but what about min_shape, opt_shape and max_shape? Let's assume we figured sparse convolution out.

Abyssaledge, 2023-09-26:
It depends on you. You should set a fix threshold to the number of points.

ArseniuML, 2023-09-26:
Is it possible to deploy FSDv2 on TensorRT with the help of SparseConv3dLayerPlugin from here:

https://github.com/jingyue202205/SE-SSD-AI-TRT

How difficult will it be to do this? Will it be possible to do this (semi) automatically, or it will mean completely rewrite FSDv2 on C++ by hands?

Abyssaledge, 2023-12-22:
The difficulty of FSDv2 deployment lies in it using spconv. So if you can find a solution for the spconv-based deployment, it should also work for FSDv2.

closed 2023-12-22 by Abyssaledge