tusen-ai/SST #163: How to choose the score threshold of the seed prediction for fsd++?

eagle-chase, 2023-09-22:
I have trained a fsdv2 model to generate seed predictions for fsd++, but I don't know how to choose the threshold of detection score. I just chose a score with better visualization results. Is it better to increase the threshold to reduce false positives, or to lower the threshold to increase the recall rate？Have you ever do some experiments？

eagle-chase, 2023-09-22:
And I found that there is a problem with the front and rear of the car being reversed during detection on my own dataset. Is there a targeted solution to this problem , or can it only be solved by expanding the data scale.

Abyssaledge, 2023-09-26:
It is hard to say what threshold is better, while I prefer to slightly a higher threshold to avoid too many false positives.
Is the heading flip frequent? I guess occasionally flip can be fixed by tracking.

closed 2023-12-22 by Abyssaledge