tusen-ai/SST #168: how to generate kitti style train.txt file

SakuraRiven, 2023-10-13:
Hi, we try to use ```python tools/create_data.py``` but found that it needs txt files such as  ```ImageSets/train.txt``` in ```kitti_format``` dir.

How could we generate these txt files so that we can reproduce the CTRL work?

Thanks for your kind help : )

SakuraRiven, 2023-10-13:
Our method: collect the ```idxs``` from ```idx2contextname.pkl``` and assign idxs to train/val/test in terms of their corresponding context_names

Abyssaledge, 2023-10-20:
Sorry for the late reply. We have been quite busy lately. Your solution seems nice, did it work? I can upload these files if you need.

SakuraRiven, 2023-10-21:
@Abyssaledge  It works well and we have reproduced the pipeline successfully : )

closed 2023-10-25 by SakuraRiven