tusen-ai/SST #169: Could you share the results of CTRL base detector？

JuiceLemonLemon, 2023-10-17:
And the performance of adopted base detector is:

OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1: [mAP 0.82786] [mAPH 0.823132]
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2: [mAP 0.751349] [mAPH 0.746877]
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1: [mAP 0] [mAPH 0]
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2: [mAP 0] [mAPH 0]
OBJECT_TYPE_TYPE_SIGN_LEVEL_1: [mAP 0] [mAPH 0]
OBJECT_TYPE_TYPE_SIGN_LEVEL_2: [mAP 0] [mAPH 0]
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1: [mAP 0] [mAPH 0]
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2: [mAP 0] [mAPH 0]
RANGE_TYPE_VEHICLE_[0, 30)_LEVEL_1: [mAP 0.93624] [mAPH 0.932365]
RANGE_TYPE_VEHICLE_[0, 30)_LEVEL_2: [mAP 0.925796] [mAPH 0.921951]
RANGE_TYPE_VEHICLE_[30, 50)_LEVEL_1: [mAP 0.819024] [mAPH 0.813976]
RANGE_TYPE_VEHICLE_[30, 50)_LEVEL_2: [mAP 0.761934] [mAPH 0.757116]
RANGE_TYPE_VEHICLE_[50, +inf)_LEVEL_1: [mAP 0.668922] [mAPH 0.661151]
RANGE_TYPE_VEHICLE_[50, +inf)_LEVEL_2: [mAP 0.542161] [mAPH 0.535483]

---------------------
@Abyssaledge 
Hello, Good morning.
Above is your shared CTRL base detector results.
And could you please share me the results of other classes (ped, cyclist)?
and the detector results, I fould you just shared the detector+ImmortalTracker results？

Abyssaledge, 2023-10-20:
Sorry for the late reply. I will check it tomorrow.

JuiceLemonLemon, 2023-10-26:
@Abyssaledge 
Hello, could you please share the fsd_base_cyc_trainset.bin?

Abyssaledge, 2023-12-22:
Sorry for the late reply, the requested file has been uploaded.

closed 2023-12-22 by Abyssaledge