tusen-ai/SST #177: Question about Argoverse validation dataset order

seonhoon1002, 2023-12-23:
Hello, I have a question regarding the order of the Argoverse validation dataset

When I visualize the Argo dataset processed by your code, the order seems shuffled(other datasets are not shuffled)

Below is a visualization result of Argoverse
[image]

Below is a visualization result of Waymo 
[image]

Could you please help me with this issue?

Abyssaledge, 2023-12-24:
I didn't shuffle it on purpose. I believe you can get the ordered results by sorting the resulted pkl file.

seonhoon1002, 2023-12-26:
In Waymo and Nuscense case, their order is correct, yes, so that's why we can use "sweep".

So I have a question about this: does your data processing process involve sorting the order in Waymo or Nuscense case?

closed 2024-01-26 by seonhoon1002