tusen-ai/SST #178: Using the Ground truth boxes as a history offline predictions in FSD++

rXYZkit, 2024-01-12:
Thank you very much for your work.  

Have you tried using the Ground truth boxes as a history offline predictions  versus using the predictions of a well-trained fsd model as a historical prediction even if you mention in your paper that the distribution gap between ground
truth used in training and predicted boxes.

Abyssaledge, 2024-01-18:
No, I have not tried it. I believe there will be a performance gap if using the GT, which may not be very significant due to the well-trained FSD having high-quality predictions. Moreover, you can narrow the gap by randomly jittering the GT.

rXYZkit, 2024-01-18:
Thanks,

The predictions of a trained model will still contain FP and TN, which bboxes are not as accurate as gt produces for generating historical skeleton points when training.  You mentioned using random jittering for gt, is that because less accurate skeleton points are more friendly to training?