tusen-ai/SST #180: Hello, could you please provide us with a model that can be used?

L00qqq, 2024-01-17:
Hello, could you please provide us with a model（ctrl） that can be used?

Abyssaledge, 2024-01-18:
We provide a link in `ctrl_instructions.md`. Has it expired?

L00qqq, 2024-01-18:
Sorry, I am a newbie in this field. You know that, the waymo dataset is very large. In fact, I just want to use this model to see the detection and tracking effect of continuous point clouds. How can I most conveniently use the pre-training you provided? How about testing the model on point cloud data? thank you very much

Abyssaledge, 2024-01-19:
If you only want to see the results, you can use the results files ('ctrl_results_in_val_and_test') in the google drive.

L00qqq, 2024-01-19:
Thank you for your reply. In fact, I want to use my local data for inference through the pre-trained model you gave me. Can I do this?