tusen-ai/SST #182: What is the goal of IngroupIndicesFunction?

ArseniuML, 2024-02-05:
A I can see, IngroupIndicesFunction from sst_ops.py is a random number generator. Why do you use it? I can imagine, that RNG can be useful for training, but you use it in the inference code...

I replaced
```
if len(torch.unique(batch_idx[fg_mask])) < batch_size:
    one_random_pos_per_sample = self.get_sample_beg_position(batch_idx, fg_mask)
    fg_mask[one_random_pos_per_sample] = True # at least one point per sample
```
to
```
if len(torch.unique(batch_idx[fg_mask])) < batch_size:
    one_random_pos_per_sample = 0
    fg_mask[one_random_pos_per_sample] = True # at least one point per sample
```
and inference results (FSDv2 with Argo2 config) seems to be identical (batch index is always 0 in my case).

Abyssaledge, 2024-02-08:
It is designed to create a fake foreground point at the start position of each sample. So, if the inference batch_size is 1, they are the same.

jiumozhi123, 2024-12-29:
作者大大您好，关于IngroupIndicesFunction这个方法，我没有太看懂。看结果似乎是生成0-35的长数组。想问下为什么会生成0-35呢，是哪块代码控制的呢？非常感谢！