tusen-ai/SST #185: track box size in each timestamp is different, Can the size of track be the same?

werider-haitao, 2024-03-07:


werider-haitao, 2024-03-07:
I found Track Coherence Optimization (TCO) post-processing in paper, is this part code open source?

Kampffussel03, 2024-06-25:
@joeless wanted to publish last July but I think it is not public yet, right?

Joeless, 2024-06-26:
Thank you for your attention to our work. To align all the 3D sizes in the track, you can simply use the strategy of selecting the size of the highest score box. The TCO code can achieve similar results using ICP. I will open-source TCO once I finish my current work. Stay tuned.

lfxx, 2024-08-01:
> Thank you for your attention to our work. To align all the 3D sizes in the track, you can simply use the strategy of selecting the size of the highest score box. The TCO code can achieve similar results using ICP. I will open-source TCO once I finish my current work. Stay tuned.

any update on this?Thank you!