tusen-ai/SST #204: Detentions are not aligned on objects

MohamedNagyMostafa, 2024-10-26:
Hello, 

I have downloaded detections provided in Google drive on readme. For some reason, the bounding boxes are not aligned on the objects. Any advice?

The point clouds are on the vehicle coordinate.

[image]

Abyssaledge, 2024-10-27:
Hi, 
Thanks for using our code. I would like to know which file the detections come from. 
From my perspective, the bounding boxes in your demonstration seem aligned with the object. Could you give a closer viewpoint on the unaligned objects?

MohamedNagyMostafa, 2024-10-27:
Thank you for the follow-up.  I have used file named "best_val_in_paper" from this link "https://drive.google.com/drive/folders/1xn8Ne-YI4IQDmJzZG_bsSv2GDK8F0Og4".

For some reason, vehicles near the AV have correct z-coordinate bounding boxes, but the boxes are wider than the objects. On the other hand, distant vehicles has bounding box above them!

[image]
[image]
[image]

Abyssaledge, 2024-10-27:
I see. Did you apply any coordinate transformation on point cloud or bounding boxes?

MohamedNagyMostafa, 2024-10-27:
I didn't do any transformation on the bounding boxes. For the point cloud, I projected it into vehicle coordinates, as in the code below. I think the detections are on vehicle frame as well, am I right?

Regarding the evaluation, I did not submit it honestly, but I believe it will get good results. I am just not sure what I did wrong to have this representation.

This is the lidar code I used to obtain 3D point cloud from range images. (for one timestamp)

```
lidar = v2.LiDARComponent.from_dict(r)
        calib = v2.LiDARCalibrationComponent.from_dict(r)
        if lidar.key.laser_name != 1:
            continue

        lidar_extrinsic = np.array(calib.extrinsic.transform).reshape(4,4)
        lidar_beam_incli_min = calib.beam_inclination.min
        lidar_beam_incli_max = calib.beam_inclination.max
        print(lidar_extrinsic)
        if len(lidar.range_image_return1.values) > 0:
            point_cloud_range_image = lidar.range_image_return1.values
            [beams, out, c] = lidar.range_image_return1.shape
            point_cloud_range_image = point_cloud_range_image.reshape((beams, out, c))
            point_cloud_ri = np.copy(point_cloud_range_image[:, :, 0])
            point_cloud_ri[point_cloud_ri < 0] = 0.0

            # Compute vertical beam inclinations.
            height = point_cloud_ri.shape[0]
            inclinations = np.linspace(lidar_beam_incli_min, lidar_beam_incli_max, height)
            inclinations = np.flip(inclinations)

            # Compute horizontal azimuth angle, correct it so that the range image center is aligned to the x-axis.
            width = point_cloud_ri.shape[1]
            az_correction = math.atan2(lidar_extrinsic[1, 0], lidar_extrinsic[0, 0])
            azimuth = np.linspace(np.pi, -np.pi, width) - az_correction

            # Sampling the range image.
            azimuth = np.broadcast_to(azimuth[np.newaxis, :], (height, width))
            inclinations = np.broadcast_to(inclinations[:, np.newaxis], (height, width))

            # Perform coordainte conversion.
            x = np.cos(azimuth) * np.cos(inclinations) * point_cloud_ri
            y = np.sin(azimuth) * np.cos(inclinations) * point_cloud_ri
            z = np.sin(inclinations) * point_cloud_ri

            # # transform point cloud into vehicle coordinate
            xyz_sensor = np.stack([x, y, z, np.ones_like(z)])
            xyz_vehicle = np.einsum('ij,jkl->ikl', lidar_extrinsic, xyz_sensor)
            xyz_vehicle = xyz_vehicle.transpose(1, 2, 0)
            xyz_vehicle = np.concatenate((xyz_vehicle[:, :, :3], point_cloud_range_image[:, :, 1, np.newaxis]), axis=2)
            pcl = xyz_vehicle[point_cloud_ri > 0]

            pcl = pcl.astype(np.float32)
            pcl.tofile(os.path.join(output_dir, f"{lidar.key.frame_timestamp_micros}.bin"))
```

Abyssaledge, 2024-10-27:
Could you give me the timestamp of the frame in your demonstration? I will upload the point cloud in my device and you can compare the two point clouds.

MohamedNagyMostafa, 2024-10-27:
Sure. Segment: "9579041874842301407_1300_000_1320_000", timestamp "1557544980097362". I also attached the point cloud I am testing on for this timestamp ( https://drive.google.com/file/d/1zVl_mP1vLDL_YpgpR1pYf2OthOpzqZ3o/view?usp=sharing ). The distant parked cars infront of the AV have their bounding boxes above them.

Abyssaledge, 2024-10-27:
You can compare your parsed point cloud with this one: https://drive.google.com/file/d/19V7_Y4hX-lV1POZfyraSkQbSGZ_HDakz/view?usp=drive_link

I noticed you use WOD v2 API.  I am not sure if it causes the problem. Look forward your comparing results.

MohamedNagyMostafa, 2024-10-27:
It is perfectly aligned with your point cloud! I am not sure if WOD v2 has a different setup. I notice your point cloud has more points compared to mine as well.

[image]

closed 2024-10-29 by MohamedNagyMostafa